kraj/meta-altera #2: perf and lttng recipes fail in YP1.8

nordoff, 2015-06-26:
When building using this layer, both u-boot and the kernel (linux-altera-ltsi-rt) build successfully, but recipes that depend on the kernel source (in my case, lttng and perf) fail to build. It appears the linux-altera-* kernel recipes are somehow not populating the build/tmp/work-shared/kernel-source directory.

```
ERROR: Function failed: do_configure (log file is located at /home/bfleming/soc_build/poky/build/tmp/work/cyclone5-poky-linux-gnueabi/perf/1.0-r9/temp/log.do_configure.27497)
ERROR: Logfile of failure stored in: /home/bfleming/soc_build/poky/build/tmp/work/cyclone5-poky-linux-gnueabi/perf/1.0-r9/temp/log.do_configure.27497
Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: Executing shell function do_configure
| sed: can't read /home/bfleming/soc_build/poky/build/tmp/work-shared/cyclone5/kernel-source/tools/perf/Makefile: No such file or directory
| WARNING: /home/bfleming/soc_build/poky/build/tmp/work/cyclone5-poky-linux-gnueabi/perf/1.0-r9/temp/run.do_configure.27497:1 exit 2 from
|   sed -i 's,-Werror ,,' /home/bfleming/soc_build/poky/build/tmp/work-shared/cyclone5/kernel-source/tools/perf/Makefile
| ERROR: Function failed: do_configure (log file is located at /home/bfleming/soc_build/poky/build/tmp/work/cyclone5-poky-linux-gnueabi/perf/1.0-r9/temp/log.do_configure.27497)
ERROR: Task 1619 (/home/bfleming/soc_build/poky/meta/recipes-kernel/perf/perf.bb, do_configure) failed with exit code '1'
ERROR: Function failed: do_make_scripts (log file is located at /home/bfleming/soc_build/poky/build/tmp/work/cyclone5-poky-linux-gnueabi/lttng-modules/2.6.0-r0/temp/log.do_make_scripts.30888)
ERROR: Logfile of failure stored in: /home/bfleming/soc_build/poky/build/tmp/work/cyclone5-poky-linux-gnueabi/lttng-modules/2.6.0-r0/temp/log.do_make_scripts.30888
Log data follows:
| DEBUG: Executing shell function do_make_scripts
| make: Entering directory `/home/bfleming/soc_build/poky/build/tmp/work-shared/cyclone5/kernel-source'
| make: *** No rule to make target `scripts'.  Stop.
| make: Leaving directory `/home/bfleming/soc_build/poky/build/tmp/work-shared/cyclone5/kernel-source'
| WARNING: /home/bfleming/soc_build/poky/build/tmp/work/cyclone5-poky-linux-gnueabi/lttng-modules/2.6.0-r0/temp/run.do_make_scripts.30888:1 exit 2 from
|   make CC="arm-poky-linux-gnueabi-gcc " LD="arm-poky-linux-gnueabi-ld.bfd " AR="arm-poky-linux-gnueabi-ar " -C /home/bfleming/soc_build/poky/build/tmp/work-shared/cyclone5/kernel-source O=/home/bfleming/soc_build/poky/build/tmp/work-shared/cyclone5/kernel-build-artifacts scripts
| ERROR: Function failed: do_make_scripts (log file is located at /home/bfleming/soc_build/poky/build/tmp/work/cyclone5-poky-linux-gnueabi/lttng-modules/2.6.0-r0/temp/log.do_make_scripts.30888)
ERROR: Task 1580 (/home/bfleming/soc_build/poky/meta/recipes-kernel/lttng/lttng-modules_2.6.0.bb, do_make_scripts) failed with exit code '1'
```

dwesterg, 2016-03-08:
This should work now.

kraj, 2016-03-11:
@dwesterg can you confirm if this is no more an issue ?

dwesterg, 2016-03-11:
work-shared/cyclone5/kernel-source is properly populated and i built lttng-ust without issue.

closed 2016-03-11 by kraj