kraj/meta-altera #11: DE1-SoC

norxander, 2016-03-11:
I have the DE1-SoC board, which generated dtb should I use? I tried the socfpga_cyclone5_socdk but when booting the system using the kernel 4.1-ltsi it got stuck on "Starting kernel ...". See full messages from botting:

7007204 bytes read in 1252 ms (5.3 MiB/s)
reading zImage
1198032 bytes read in 217 ms (5.3 MiB/s)
reading socfpga.dtb
19526 bytes read in 8 ms (2.3 MiB/s)
## Starting application at 0x3FF79528 ...
## Application terminated, rc = 0x0
## Flattened Device Tree blob at 00000100
   Booting using the fdt blob at 0x00000100
   reserving fdt memory region: addr=0 size=1000
   Loading Device Tree to 03ff8000, end 03fffc45 ... OK

Starting kernel ...

Any advice? 
Thanks.

kraj, 2016-03-12:
can you test wth latest tip and see if it helps ?

norxander, 2016-03-12:
What and where are the tips?

kraj, 2016-03-12:
I meant latest top of the tree

norxander, 2016-03-12:
Actually yesterday I updated all the trees to get the kernel 4.1 LTSI, and gcc 5.2 from the meta linaro, and after building the filesystem, kernel and dtb file the card got stuck in that point I mentioned.
Before that I was using the kernel 3.10 LTSI and it was working just fine.

dwesterg, 2016-03-14:
The dts for the board you are using is not included in the kernel.  You will need to create one, and patch it in or compile a dtb out of tree.  I dont have one of those boards so i cannot test it.  In the 3.10ltsi kernel, you were using the socfpga_cyclone5_socdk devicetree on this board?  are you certain you updated the dtb on the fat partition of the sdcard?

norxander, 2016-03-14:
In 3.10 ltsi kernel I was using the one provided by the kernel (socfpga_cyclone5.dtb), for kernel 4.1 ltsi there is one named as socfpga_cyclone5_socdk. The kernel 3.10 ltsi was working fine for me using the  dtb provided by the kernel tree (socfpga_cyclone5) but after updating to get the kernel 4.1 ltsi and compiling it with the linaro gcc 5.2 and got 3 different dtbs, I tried the socfpga_cyclone5_socdk.dtb and it didn't work. I supposed it was because of the dtb file, because of the messages I placed in the first comment. I will build the kernel, dtb and uboot again using the latest commit just to make sure I didn't update a file in the sd card. And will let you know guys.

dwesterg, 2016-03-15:
I will have to look at the dtbs, but some newer board features were enabled
in the default cyclone5 sdk dtb and so it may not work on the de1 board.
 you may just need to remove a few features.

kraj, 2016-03-15:
@dwesterg I am inclined to keep 3.10 recipes in there, since 3.10 is still a LTSI kernel which is supported and it seems not all altera boards may be able to cope with new kernels as it seems, until its all settled.

dwesterg, 2016-03-15:
yup, i was just working on that.  I have the 3.10 kernels working if using
the 4.9 toolchain.  I am just trying to figure out how to force the
toolchain version for a kernel recipe.

kraj, 2016-03-15:
Dont mix and match toolchains. OE will not allow it. You can pin to jethro branch of OE-Core and set GCCVERSION = "4.9%" in your local.conf

dwesterg, 2016-03-15:
Do you think it makes sense to start branching off the yocto/angtrom versions in the repo?  it would make it a bit easier for yocto users to follow.

kraj, 2016-03-16:
please use jethro branch for now.

closed 2016-03-16 by kraj