kraj/meta-altera #30: odd test in classes/image_types_socfpga.bbclass

gizero, 2016-06-27:
I had a couple of glitches with the sdcard class I will summarize here, for the sake of pointing them out and seeking comments.

1. We should probably provide symlinks to the latest sdcard image artifact we build, similarly to what the framework already is doing with other image types. I have a patch for this I'll submit as soon as polished enough.
1. Don't know the reason yet, but I have to re-apply the `set part 3 to type a2 for spl / uboot image` thing on the generated image to make it bootable on the target. I did no further investigation after finding the workaround... Any clue here?
1. While looking at the class code, I noticed the conditional code, quoted below, being odd since both conditions in the `if` and `elif` statements are the same. I guess the `elif` one was supposed to be `elif [ -e "${DEPLOY_DIR_IMAGE}/${SPL_BINARY}" ]` instead (without the `-${UBOOT_CONFIG}`). If so, I can provide a patch as well.

```
generate_28nm_sdcard () {
[...]
	if [ -e "${DEPLOY_DIR_IMAGE}/${SPL_BINARY}-${UBOOT_CONFIG}" ]
	then
            dd if=${DEPLOY_DIR_IMAGE}/${SPL_BINARY}-${UBOOT_CONFIG} of=${SDCARD} conv=notrunc seek=1 bs=$(expr 1024 \* 1024)
	elif [ -e "${DEPLOY_DIR_IMAGE}/${SPL_BINARY}-${UBOOT_CONFIG}" ]
	then
            dd if=${DEPLOY_DIR_IMAGE}/${SPL_BINARY} of=${SDCARD} conv=notrunc seek=1 bs=$(expr 1024 \* 1024)
	else
            bbfatal "${SPL_BINARY} does not exist."
fi
```

dwesterg, 2016-06-27:
This is still a work in progress, it should probably be removed when we branch off krogoth.

Thanks for the feedback

--dalon

kraj, 2017-01-12:
krogoth is already branched out. reopen if still a problem.

closed 2017-01-12 by kraj