kraj/meta-altera #42: u-boot-socfpga_2014.10 broken with poky master (arria10)

gizero, 2016-11-01:
While crafting #41 I checked if the same issue applied to `u-boot-socfpga_2014.10` also, which is the preferred version for `arria10` machine. This is not the case since it re-defines all the variable that have been moved out of `u-boot.inc`, but I noticed that `u-boot-socfpga_2014.10` is broken anyway. I did some bisecting on poky repo and it seems that the commit to blame is `239efccccaa74912b3fc9ec717e93f528e2c3836` which bumps  `GCCVERSION`. I'm not very familiar with `arria10` and the error gcc is throwing, hence I'm filing this ticket for people more involved.

kraj, 2016-11-01:
can you post the error message that you are seeing ?

gizero, 2016-11-01:
This is the log when baking `virtual/bootloader`...

```
Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "arria10"
DISTRO            = "poky"
DISTRO_VERSION    = "2.2"
TUNE_FEATURES     = "arm armv7a vfp  neon"
TARGET_FPU        = "softfp"
meta
meta-poky
meta-yocto-bsp    = "master:927ed6cbefbaea2905919e158ceda9546e3e710e"
meta-altera       = "master:c56e9935de14385e03e8af6a3221b1f158ce4f0a"

Initialising tasks: 100% |###################################################| Time: 0:00:00
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: u-boot-socfpga-2014.10-r0 do_compile: oe_runmake failed
ERROR: u-boot-socfpga-2014.10-r0 do_compile: Function failed: do_compile (log file is located at /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/
2014.10-r0/temp/log.do_compile.25668)
ERROR: Logfile of failure stored in: /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/temp/log.do_compile.25668
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 1 CROSS_COMPILE=arm-poky-linux-gnueabi- CC=arm-poky-linux-gnueabi-gcc  --sysroot=/home/gizero/work/upstreaming/build2/tmp/sysroots/arria10 V=1 HOSTCC=gcc  -isystem/home/
gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -L/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/lib -L/home/gizero/work/upstreaming/b
uild2/tmp/sysroots/x86_64-linux/lib -Wl,-rpath-link,/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath-link,/home/gizero/work/upstreaming/build2/tmp/sysro
ots/x86_64-linux/lib -Wl,-rpath,/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath,/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/lib -Wl,
-O1 -C /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/git O=/home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/
u-boot-socfpga/2014.10-r0/build socfpga_arria10_defconfig
| NOTE: make -j 1 CROSS_COMPILE=arm-poky-linux-gnueabi- CC=arm-poky-linux-gnueabi-gcc  --sysroot=/home/gizero/work/upstreaming/build2/tmp/sysroots/arria10 V=1 HOSTCC=gcc  -isystem/home/
gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -L/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/lib -L/home/gizero/work/upstreaming/b
uild2/tmp/sysroots/x86_64-linux/lib -Wl,-rpath-link,/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath-link,/home/gizero/work/upstreaming/build2/tmp/sysro
ots/x86_64-linux/lib -Wl,-rpath,/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath,/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/lib -Wl,
-O1 -C /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/git O=/home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/
u-boot-socfpga/2014.10-r0/build all
| make: Entering directory `/home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/git'
| make -C /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/build \
|       KBUILD_SRC=/home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/git \
|       KBUILD_EXTMOD="" -f /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/git/Makefile \
|       socfpga_arria10_defconfig
| make -f /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/git/scripts/Makefile.build obj=scripts/basic
|   gcc  -isystem/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -L/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/lib -L/home/gi
zero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/lib -Wl,-rpath-link,/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath-link,/home/gizero/work/upstr
eaming/build2/tmp/sysroots/x86_64-linux/lib -Wl,-rpath,/home/gizero/work/upstreaming/build2/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath,/home/gizero/work/upstreaming/build2/tmp/sysroot
s/x86_64-linux/lib -Wl,-O1 -Wp,-MD,scripts/basic/.fixdep.d -Iscripts/basic -Wall -Wstrict-prototypes -O2 -fomit-frame-pointer -o scripts/basic/fixdep /home/gizero/work/upstreaming/build
2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/git/scripts/basic/fixdep.c
| rm -f .tmp_quiet_recordmcount
| ln -fsn /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/git source
| /bin/bash /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/git/scripts/mkmakefile \
|           /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/git /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnuea
bi/u-boot-socfpga/2014.10-r0/build 2014 10
|   GEN     /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/build/Makefile
| /bin/bash /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/git/scripts/multiconfig.sh socfpga_arria10_defconfig
[... cut ...]
| ERROR: Function failed: do_compile (log file is located at /home/gizero/work/upstreaming/build2/tmp/work/arria10-poky-linux-gnueabi/u-boot-socfpga/2014.10-r0/temp/log.do_compile.25668
)
ERROR: Task (/home/gizero/work/upstreaming/meta-altera/recipes-bsp/u-boot/u-boot-socfpga_2014.10.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 393 tasks of which 387 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/gizero/work/upstreaming/meta-altera/recipes-bsp/u-boot/u-boot-socfpga_2014.10.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

dwesterg, 2017-01-11:
I have replicated this issue, the v2014.10 version of uboot does not currently support GCC6.  It may be possible to patch this in as at this time arria10 is not supported in mainline uboot.  For yocto I would suggest  just setting your preferred compiled version to 5.3.

--dalon

kraj, 2017-01-12:
Can be fixed with patch like https://github.com/Freescale/meta-fsl-arm/blob/master/recipes-bsp/u-boot/u-boot-ls1/fix-build-error-under-gcc6.patch

I will be willing to accept such a patch into layer.

dwesterg, 2017-01-14:
fixed

closed 2017-01-19 by kraj