kraj/meta-altera #78: rocko virtual/kernel unable to find revision b84195...a451d1 in branch socfpga-4.1.33-ltsi even from upstream

bfurmanAtBird, 2018-03-08:
Attempting to build meta-altera/rocko with poky/rocko:

bran@Bds2Ubu14:/media/bran/WORKSPACE/rocketboards/rocko/build$ bitbake virtual/kernel
WARNING: Host distribution "ubuntu-14.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |########################################################################################| Time: 0:00:00
Loaded 1294 entries from dependency cache.
Parsing recipes: 100% |######################################################################################| Time: 0:00:00
Parsing of 836 .bb files complete (835 cached, 1 parsed). 1295 targets, 75 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal-4.8"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "arria10"
DISTRO               = "poky"
DISTRO_VERSION       = "2.4.2"
TUNE_FEATURES        = "arm armv7a vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "rocko:5ba69a97ab5faa8f3866aaeab1d6eaa3cb8149ed"
meta-altera          = "rocko:d671a199d5d3ad820901121eb53761471b8fba41"

Initialising tasks: 100% |###################################################################################| Time: 0:00:00
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: linux-altera-ltsi-4.1.33-ltsi+gitAUTOINC+b84195c056-r0 do_fetch: Failed to fetch URL git://github.com/altera-opensource/linux-socfpga.git;protocol=https;branch=socfpga-4.1.33-ltsi, attempting MIRRORS if available
ERROR: linux-altera-ltsi-4.1.33-ltsi+gitAUTOINC+b84195c056-r0 do_fetch: Fetcher failure: Unable to find revision b84195c056ee943efa9eff2767b09dd266a451d1 in branch socfpga-4.1.33-ltsi even from upstream
ERROR: linux-altera-ltsi-4.1.33-ltsi+gitAUTOINC+b84195c056-r0 do_fetch: Fetcher failure for URL: 'git://github.com/altera-opensource/linux-socfpga.git;protocol=https;branch=socfpga-4.1.33-ltsi'. Unable to fetch URL from any source.
ERROR: linux-altera-ltsi-4.1.33-ltsi+gitAUTOINC+b84195c056-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /media/bran/WORKSPACE/rocketboards/rocko/build/tmp/work/arria10-poky-linux-gnueabi/linux-altera-ltsi/4.1.33-ltsi+gitAUTOINC+b84195c056-r0/temp/log.do_fetch.22257
ERROR: Task (/media/bran/WORKSPACE/rocketboards/rocko/meta-altera/recipes-kernel/linux/linux-altera-ltsi_4.1.33.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 127 tasks of which 109 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /media/bran/WORKSPACE/rocketboards/rocko/meta-altera/recipes-kernel/linux/linux-altera-ltsi_4.1.33.bb:do_fetch
Summary: There were 2 WARNING messages shown.
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.

dl9pf, 2018-03-09:
Yes, same here:
WARNING: linux-altera-ltsi-4.1.33-ltsi+gitAUTOINC+b84195c056-r0 do_fetch: Failed to fetch URL git://github.com/altera-opensource/linux-socfpga.git;protocol=https;branch=socfpga-4.1.33-ltsi, attempting MIRRORS if available
DEBUG: linux-altera-ltsi-4.1.33-ltsi+gitAUTOINC+b84195c056-r0 do_fetch: Fetcher failure: Unable to find revision b84195c056ee943efa9eff2767b09dd266a451d1 in branch socfpga-4.1.33-ltsi even from upstream

dl9pf, 2018-03-09:
https://github.com/altera-opensource/linux-socfpga

does no longer contain a socfpga-4.1.33-ltsi branch. 

Only socfpga-4.9.78-ltsi , tip is e681fba61a85fa13e546bba277083c49c091e361 .

kraj, 2018-03-09:
I have notified Altera about this.

dl9pf, 2018-04-03:
Any response ?

kraj, 2018-04-05:
Nope. it might be an option to switch to 

https://github.com/kraj/linux-socfpga/commits/socfpga-4.1-ltsi

multiple1, 2019-10-12:
I think we can solve the issue by not including the branch in the git fetch step.
The commit rev for the release (TAG) is still there without branch:
https://github.com/altera-opensource/linux-socfpga/releases/tag/rel_socfpga-4.14.73-ltsi-rt_19.07.02_pr

KBRANCH ?= "${LINUX_VERSION_PREFIX}${LINUX_VERSION}${LINUX_VERSION_SUFFIX}"
SRC_URI = "${KERNEL_REPO};protocol=${KERNEL_PROT};branch=${KBRANCH}"

kraj, 2019-12-04:
rocko is not maintained anymore.

closed 2019-12-04 by kraj

Deliver88, 2021-03-18:
> Nope. it might be an option to switch to
> 
> https://github.com/kraj/linux-socfpga/commits/socfpga-4.1-ltsi

Hello,

I faced same error, 

linux-altera-ltsi-4.1.33-ltsi+gitAUTOINC+b84195c056-r0 do_fetch: Fetcher failure: Unable to find revision b84195c056ee943efa9eff2767b09dd266a451d1 in branch socfpga-4.1.33-ltsi even from upstream

Where (file/files) should I change the branch "socfpga-4.1.33-ltsi" to  this https://github.com/kraj/linux-socfpga/commits/socfpga-4.1-ltsi one?  I am following this guide https://github.com/intel/meta-de10-nano 

Moreover, how could I switch e.g. to 4.14 ltsi for de10-nano? which one is recommended to use?

Thanks.