kraj/meta-altera #93: Fetch Error of 4.11

jyu0015, 2018-07-30:
I was trying to build Angstrom for SoCFPGA. The building process finished with a error, and it seems like a link was disconnected. How can I fix the error?

NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Running task 514 of 2237 (/home/jyu/layers/meta-altera/recipes-kernel/linux/linux-altera_4.11.bb:do_fetch)
NOTE: Running task 548 of 2237 (virtual:native:/home/jyu/layers/openembedded-core/meta/recipes-core/libxml/libxml2_2.9.4.bb:do_install)
NOTE: recipe linux-altera-4.11+gitAUTOINC+d0851166ca-r0: task do_fetch: Started
NOTE: recipe libxml2-native-2.9.4-r0: task do_install: Started
WARNING: linux-altera-4.11+gitAUTOINC+d0851166ca-r0 do_fetch: Failed to fetch URL git://github.com/altera-opensource/linux-socfpga.git;protocol=https;branch=socfpga-4.11, attempting MIRRORS if available
NOTE: recipe libxml2-native-2.9.4-r0: task do_install: Succeeded
ERROR: linux-altera-4.11+gitAUTOINC+d0851166ca-r0 do_fetch: Fetcher failure: Unable to find revision d0851166caa4ba9f6e2923ffa37f476ebb99e9a0 in branch socfpga-4.11 even from upstream
ERROR: linux-altera-4.11+gitAUTOINC+d0851166ca-r0 do_fetch: Fetcher failure for URL: 'git://github.com/altera-opensource/linux-socfpga.git;protocol=https;branch=socfpga-4.11'. Unable to fetch URL from any source.
ERROR: linux-altera-4.11+gitAUTOINC+d0851166ca-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /home/jyu/build/tmp-angstrom-glibc/work/cyclone5-angstrom-linux-gnueabi/linux-altera/4.11+gitAUTOINC+d0851166ca-r0/temp/log.do_fetch.25173
NOTE: recipe linux-altera-4.11+gitAUTOINC+d0851166ca-r0: task do_fetch: Failed
NOTE: Running task 549 of 2237 (virtual:native:/home/jyu/layers/openembedded-core/meta/recipes-core/libxml/libxml2_2.9.4.bb:do_populate_sysroot)

kraj, 2018-08-02:
@dalon can you take a look

dwesterg, 2018-09-04:
4.11 is no longer supported and has been removed from the altera linux fork.  Altera only maintains the latest kernel, latest kernel -1, and the current LTSI kernel. 

It is strongly suggested, should you need to deviate from these supported versions,  that you mirror / clone the linux-socfpga repo.

--dalon

kraj, 2019-07-08:
thanks @dwesterg

closed 2019-07-08 by kraj