kraj/meta-altera #99: Angstrom wic image not booting on DE0-Nano-SOC Board / Cyclone-V

feddischson, 2018-10-19:
Hi,

I was trying to get a Angstrom wic image running, including
kraj/meta-altera (786bee6), but facing the issue that the sd-card image
is not booting. It seems that the pre-loader is not getting executed.

And I also discovered, that the vfat partition does not contain a
u-boot.img.

For doing this, I am using a Terasic DE0-nano-soc board, which includes
a Cyclone-V SOC.

To narrow the issue a little bit: 
I replaced `deploy/glibc/images/cyclone5/u-boot-with-spl.sfp` by a manual binary created via Intel/Atlera's `bsp-generate-files` -> `preloader-mkpimage.bin`

After replacing `u-boot-with-spl.sfp` and providing `u-boot.img` on the vfat partition, the system was booting fine.

dwesterg, 2018-10-19:
Thw wic uses mainline uboot, it does not put a u-boot.img in the fat partition as mainline wraps 4 spls and a full uboot for the a2 partition.  Your issue is likely due to the DE0 nano board using early production silicon which had a known issue with clock configuration.  I do not believe mainline uboot deals with this, but these patches should address those.

https://github.com/intel/meta-de10-nano/tree/master/recipes-bsp/u-boot/files/v2017.03

--dalon

dwesterg, 2019-06-03:
This now works.

closed 2019-06-12 by kraj