kraj/meta-altera #153: DE0 Nano SoC doesn't boot with default Cyclone 5 machine

auserdude, 2019-12-09:
Hello, 

I have noticed that the DE0 Nano SoC board does not boot with the Cyclone 5 machine conf.
By modifying cyclone5.conf with the following values:

```
#UBOOT_CONFIG ??= "cyclone5-socdk"
UBOOT_CONFIG ??= "de0-nano-soc"
```

and 

```
#UBOOT_EXTLINUX_FDT_default ?= "../socfpga_cyclone5_socdk.dtb"
UBOOT_EXTLINUX_FDT_default ?= "../socfpga_cyclone5_de0_nano_soc.dtb"
```

the board boots normally.

This has been done with 
meta-altera commit `c8635e2b9720cfd74f96a58885628cc95276e43d` 
and 
Yocto commit `da81542171f0cab0ea1a9bf45ac3c256e6f343f9`

What about a dedicated de0_nano_soc.conf machine file to simplify the configuration? 
Or, maybe, at least, some doc/comments to explain this?

dwesterg, 2019-12-09:
The preferred method, imo, is to deal with this in the local.conf rather than having many machine confs with the same underlying hardware.

You can simply add

UBOOT_CONFIG = "de0-nano-soc"
UBOOT_EXTLINUX_FDT_default = "../socfpga_cyclone5_de0_nano_soc.dtb"

to the local.conf in your build.  I will add something to this affect in the readme.

--dalon

auserdude, 2019-12-10:
> The preferred method, imo, is to deal with this in the local.conf rather than having many machine confs with the same underlying hardware.
>
> You can simply add
>
> UBOOT_CONFIG = "de0-nano-soc" [...]

Sounds like a plan.

closed 2019-12-10 by auserdude