kraj/meta-altera #156: HPS2FPGA bridges disabled by default

auserdude, 2019-12-15:
Hello,

Not sure if this is a possible improvement or a bug.
HPS 2 FPGA bridges driver is available in the kernel but it doesn't activate the bridges.

If you look at the [kernel driver](https://github.com/torvalds/linux/blob/1522d9da40bdfe502c91163e6d769332897201fa/drivers/fpga/altera-hps2fpga.c#L67) for the HPS to FPGA bridges you can see that the function expects an "enable" parameter. It seems to be triggered by the device tree property `bridge-enable`.

I have realized there are 2 devices in the `socfpga_cyclone5_de0_nano_soc.dts` file, related to the HPS-to-FPGA bridges (the normal one and the lightweight one). These two devices don't have the "bridge-enable" property at all. Not even in the Altera Linux repository.

Is there any chance to activate the bridges by default with Meta-Altera or to provide a way to facilitate this operation?

dwesterg, 2019-12-16:
There is no expectation or requirement that the FPGA be programmed prior to booting linux, so the dts entry for the bridges is intentionally left blank so that whatever u-boot enables is left alone.  The expectation is that a user applies a dts overlay to add fpga peripherals and enable the bridges when necessary, OR program and enable the bridges in u-boot.  

The only safe action is what we currently do.  Enabling the bridges by default could be problematic from unconfigured FPGAs.

-dalon

auserdude, 2019-12-16:
Ok, I agree this is the safest approach.
These are my first experiences with Linux + FPGA so I'm trying to understand the best way to turn on the bridges when necessary. Do you suggest to activate the bridges when the peripherals' drivers  (for those peripherals that use the bridges) are initialized?

closed 2019-12-23 by auserdude