kraj/meta-altera #164: stratix10: DDR: Error as SDRAM calibration failed

btgoodwin, 2020-02-07:
We're trying to implement a warrior -based yocto build since it's suggested as the latest successful build [here on rocketboards](https://rocketboards.org/foswiki/view/Documentation/YoctoDoraBuildWithMetaAltera).  We are using poky's warrior branch and the master branch of meta-altera from here.

The build succeeds without error, but upon boot-up, it presents this message:

```
U-Boot SPL 2017.09 (Sep 22 2018 - 07:29:05)
MPU         1000000 kHz
L3 main     400000 kHz
Main VCO    2000000 kHz
Per VCO     2000000 kHz
EOSC1       25000 kHz
HPS MMC     50000 kHz
UART        100000 kHz
DDR: Initializing Hard Memory Controller
DDR: Triggerring emif_reset
DDR: emif_reset triggered successly
DDR: Triggerring emif_reset
DDR: emif_reset triggered successly
DDR: Triggerring emif_reset
DDR: emif_reset triggered successly
DDR: Error as SDRAM calibration failed
DDR: Initialization failed.
### ERROR ### Please RESET the board ###
```

Previously, using thud branches and a `bbappend` for the kernel recipe to fix a now-defunct branch name, we were able to boot fine.  It was only this switch to warrior that brought this problem out.  

We noticed the version doesn't match what was built (2019.10 on the u-boot-socfpga recipe vs. 2017.09 in that log).  The DIP switches are configured for SD card boot.

Has anyone seen this or know of a way to debug it?

dwesterg, 2020-02-07:
Hi,

I suspect you have not updated the QSPI with an update FPGA / u-boot-spl image?  2017.09 u-boot is not in the master meta-altera branch.  remember that you need to take the u-boot-spl hex and add it to the FPGA image.

https://rocketboards.org/foswiki/Documentation/S10GSRDCompilingHardwareDesign180

--dalon

btgoodwin, 2020-07-28:
I apologize for leaving this open so long.  It's OBE at this point and the team that was pursuing it is just now revisiting but has a completely different problem (hang at `SDRAM-ECC: Initialized success with 920ms`), however they're programming a combined bitstream, etc. over JTAG and are trying to find an ihex output from the yocto build process since that seems to be how they're now incorporating things manually during development.

We can close this.  If they hit the same snag, we'll re-open. I've pointed them to this item as well just in case your explanation gets them any farther.

Thanks again, Dalon!

closed 2020-07-28 by btgoodwin