kraj/meta-altera #165: linux-altera-5.4 unable to find file file://lbdaf.scc

bernardoaraujor, 2020-02-09:
I'm trying to run a build based on thud branch.
```
Parsing recipes: 100% |###################| Time: 0:00:00
Parsing of 801 .bb files complete (795 cached, 6 parsed). 1273 targets, 68 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: preferred version v2019.01% of u-boot-socfpga not available (for item virtual/bootloader)
NOTE: versions of u-boot-socfpga available: 1:v2019.04+gitAUTOINC+83e929c739 1:v2019.10+gitAUTOINC+f5efc10657
NOTE: preferred version v2019.01% of u-boot-socfpga not available (for item u-boot-socfpga-dev)
NOTE: versions of u-boot-socfpga available: 1:v2019.04+gitAUTOINC+83e929c739 1:v2019.10+gitAUTOINC+f5efc10657
NOTE: preferred version v2019.01% of u-boot-socfpga not available (for item u-boot-socfpga)
NOTE: versions of u-boot-socfpga available: 1:v2019.04+gitAUTOINC+83e929c739 1:v2019.10+gitAUTOINC+f5efc10657

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "cyclone5"
DISTRO               = "poky"
DISTRO_VERSION       = "2.6.4"
TUNE_FEATURES        = "arm armv7a vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "thud:958427e9d2ee7276887f2b02ba85cf0996dea553"
meta-altera          = "thud:823bf2bff467d7006a75be9eb8baece7f32c4973"

Initialising tasks: 100% |################| Time: 0:00:01
Sstate summary: Wanted 978 Found 0 Missed 978 Current 86 (0% match, 8% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: linux-altera-5.4+gitAUTOINC+526ec9a1b0-r0 do_fetch: Failed to fetch URL file://lbdaf.scc, attempting MIRRORS if available
ERROR: linux-altera-5.4+gitAUTOINC+526ec9a1b0-r0 do_fetch: Fetcher failure: Unable to find file file://lbdaf.scc anywhere. The paths that were searched were:
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/config/poky
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/linux-altera-5.4+gitAUTOINC+526ec9a1b0/poky
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/linux-altera/poky
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/files/poky
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/config/cyclone5
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/linux-altera-5.4+gitAUTOINC+526ec9a1b0/cyclone5
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/linux-altera/cyclone5
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/files/cyclone5
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/config/armv7a
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/linux-altera-5.4+gitAUTOINC+526ec9a1b0/armv7a
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/linux-altera/armv7a
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/files/armv7a
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/config/arm
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/linux-altera-5.4+gitAUTOINC+526ec9a1b0/arm
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/linux-altera/arm
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/files/arm
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/config/
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/linux-altera-5.4+gitAUTOINC+526ec9a1b0/
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/linux-altera/
    /home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/files/
    /home/bernardo/dev/de10-nano/build/downloads
ERROR: linux-altera-5.4+gitAUTOINC+526ec9a1b0-r0 do_fetch: Fetcher failure for URL: 'file://lbdaf.scc'. Unable to fetch URL from any source.
ERROR: linux-altera-5.4+gitAUTOINC+526ec9a1b0-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /home/bernardo/dev/de10-nano/build/tmp/work/cyclone5-poky-linux-gnueabi/linux-altera/5.4+gitAUTOINC+526ec9a1b0-r0/temp/log.do_fetch.20004
ERROR: Task (/home/bernardo/dev/de10-nano/meta-altera/recipes-kernel/linux/linux-altera_5.4.bb:do_fetch) failed with exit code '1'
```

kraj, 2020-02-12:
Can you backport https://github.com/kraj/meta-altera/commit/831a90acb1ad3d8189c887bf200d2143fb4bbff1 to thud ?
@dwesterg is this intentionally missing on thud ?

dwesterg, 2020-02-12:
> Can you backport 831a90a
> to thud ?
> @dwesterg <https://github.com/dwesterg> is this intentionally missing on
> thud ?

no, i missed it

dwesterg, 2020-02-12:
i will fix this in a separate patch, cherry-picking isnt applying cleanly.

kraj, 2020-02-12:
Thanks @dwesterg @bernardoaraujor can you update to latest thud and retry

zandrey, 2020-04-21:
Hello all!

I just stumbled across this issue here and wanted to check if this fragment is really needed? I've already submitted the patch for this [cb46b0991c273a001e73ca9cd44b067e2c596892](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=cb46b0991c273a001e73ca9cd44b067e2c596892) against mainline, so the large block device support should be enabled by default in the mainline kernel.

@kraj @dwesterg I guess you can drop this fragment from the layer.

-- andrey

dwesterg, 2020-04-21:
yes, looks that way.  Thanks.

dwesterg, 2020-04-21:
The two older kernels (4.14 ones) still require this.

kraj, 2020-04-21:
so perhaps it will be good to have a patch for this.

dwesterg, 2020-04-21:
yes, i am also thinking of just removing the older kernels as they will not
be supported for much longer in any case as the newer lts kernels have been
made available.

I will remove the lbdaf.scc from the kernels that dont need it for sure.

--dalon

kraj, 2020-04-21:
for dunfell I would interested to drop it yes.