kraj/meta-altera #174: Updated from linux-altera_5.1 to linux-altera_5.3. Unable to boot until after warm-reset

rodinia-hsandlar, 2020-03-09:
I'm using the DE-10-Nano and recently updated from the kernel recipe for 5.1 to the recipe for 5.3. After this update, I've noticed that the board no longer boots on the first attempt, but will eventually boot after a WDT timeout + soft-reset, or just by pressing the warm-reset button on this development kit.  

To verify this, I swapped in the 5.1 kernel into my 5.3 sdcard build and everything boots as expected.  

In all cases, this is after configuring the FPGA. Here are a couple example logs:
Using the 5.3 kernel:
```
U-Boot 2019.10 (Mar 09 2020 - 16:59:36 +0000)

CPU:   Altera SoCFPGA Platform
FPGA:  Altera Cyclone V, SE/A6 or SX/C6 or ST/D6, version 0x0
BOOT:  SD/MMC Internal Transceiver (3.0V)
       Watchdog enabled
DRAM:  1 GiB
MMC:   dwmmc0@ff704000: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Terasic DE-0(Atlas)
Net:
Error: ethernet@ff702000 address not set.
eth-1: ethernet@ff702000
Hit any key to stop autoboot:  0
=>
=> fatload mmc 0:2 $loadaddr soc_system.rbf
7007204 bytes read in 347 ms (19.3 MiB/s)
=> fpga load 0 $loadaddr $filesize
=> bridge enable
=> fatload mmc 0:2 $loadaddr zImage
5610728 bytes read in 275 ms (19.5 MiB/s)
=> fatload mmc 0:2 $fdt_addr_r soc_system.dtb
25692 bytes read in 5 ms (4.9 MiB/s)
=> setenv bootargs console=ttyS0,115200 root=/dev/mmcblk0p3 rw rootwait
=> bootz $loadaddr - $fdt_addr_r
## Flattened Device Tree blob at 02000000
   Booting using the fdt blob at 0x2000000
   Loading Device Tree to 09ff6000, end 09fff45b ... OK

Starting kernel ...

Deasserting all peripheral resets
```
^^ This does not continue until a warm-reset occurs. At which point it will boot fine.

Here is the 5.1 kernel boot log:
```
**U-Boot 2019.10 (Mar 09 2020 - 16:59:36 +0000)

CPU:   Altera SoCFPGA Platform
FPGA:  Altera Cyclone V, SE/A6 or SX/C6 or ST/D6, version 0x0
BOOT:  SD/MMC Internal Transceiver (3.0V)
       Watchdog enabled
DRAM:  1 GiB
MMC:   dwmmc0@ff704000: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Model: Terasic DE-0(Atlas)
Net:
Error: ethernet@ff702000 address not set.
eth-1: ethernet@ff702000
Hit any key to stop autoboot:  0
=>
=>
=> fatload mmc 0:2 $loadaddr soc_system.rbf
7007204 bytes read in 342 ms (19.5 MiB/s)
=> fpga load 0 $loadaddr $filesize
=> bridge enable
=> fatload mmc 0:2 $loadaddr zImage
4848736 bytes read in 237 ms (19.5 MiB/s)
=> fatload mmc 0:2 $fdt_addr_r soc_system.dtb
25692 bytes read in 5 ms (4.9 MiB/s)
=> setenv bootargs console=ttyS0,115200 root=/dev/mmcblk0p3 rw rootwait
=> bootz $loadaddr - $fdt_addr_r
## Flattened Device Tree blob at 02000000
   Booting using the fdt blob at 0x2000000
   Loading Device Tree to 09ff6000, end 09fff45b ... OK

Starting kernel ...

Deasserting all peripheral resets
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.1.0-altera (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 SMP Tue Aug 20 19:46:29 UTC 2019
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c5387d

Continues boot just fine ...
```

Do you have any idea as to why this change in behavior is present after the kernel update?

kraj, 2020-03-10:
@dwesterg FYI can you take a look

rodinia-hsandlar, 2020-03-12:
I'm closing this issue. This turned out to be unrelated to the kernel upgrade. 

The 5.1 kernel I was testing with did not have the framebuffer driver (opencores framebuffer - ocfb) enabled. Part of my hardware integration for the opencores framebuffer included hooking up a data fetch interface, which I had connected to an fpga2sdram bridge. 

Apparently the fpga2sdram bridge was not being enabled by u-boot or any other kernel driver prior to the ocfb driver loading, which resulted in a full system bus stall when the hardware tried to access that interface. 

I am unsure about why a warm reset seemed to 'fix' this problem. 

In any case, after moving the data fetch interface to use the f2h_axi_slave bridge this issue has been resolved.

closed 2020-03-12 by rodinia-hsandlar