kraj/meta-altera #186: Building openssh tools into filesystem for CycloneV

jackfrye, 2020-05-12:
Hello. 
I am struggling to build ssh utilities into my rootfs for CycloneV SoC.

I am following this guide.
https://rocketboards.org/foswiki/Documentation/BuildingBootloader

Specifically, I am taking these steps
```
cd $ROOTFS_TOP
rm -rf cv && mkdir cv && cd cv
git clone -b zeus git://git.yoctoproject.org/poky.git
git clone -b master git://github.com/kraj/meta-altera.git
sed -i '/socfpga_cyclone5_de0_nano_soc.dtb/d' meta-altera/conf/machine/cyclone5.conf
source poky/oe-init-build-env ./build
echo 'MACHINE = "cyclone5"' >> conf/local.conf
echo 'IMAGE_FSTYPES = "tar.gz"' >> conf/local.conf
bitbake core-image-minimal
ln -s $ROOTFS_TOP/cv/build/tmp/deploy/images/cyclone5/core-image-minimal-cyclone5.tar.gz $LINUX_BIN/a9/
```
I have also added the line
`IMAGE_INSTALL_append = " openssh-scp openssh-sftp-server openssh-ssh"
`
at the end of my local.conf

When I untar the rootfs and boot up Linux (built with the directions below)
```
cd $LINUX_TOP
git clone https://github.com/altera-opensource/linux-socfpga linux-socfpga.a9
cd linux-socfpga.a9
git checkout -b test -t origin/socfpga-5.4.23-lts
make socfpga_defconfig
make -j 24 zImage Image dtbs modules
make modules_install INSTALL_MOD_PATH=modules_install
rm -rf modules_install/lib/modules/*/build
rm -rf modules_install/lib/modules/*/source
```
I do not see anything in the /etc directory that I would expect to be there for ssh service. Have I copied the wrong file system build output? I tried to follow the guide as closely as possible with that addition to the IMAGE_INSTALL. What should I expect to see?

kraj, 2020-05-13:
Please untar core-image-minimal-cyclone5.tar.gz in a temp location and check that ssh related files are in or on. From there on its not clear what you are doing with compiling kernel etc. is it on board directly or is it cross compiled I have no idea.

jackfrye, 2020-05-13:
For whatever reason, when I followed this guide:
https://wiki.yoctoproject.org/wiki/Cookbook:Example:Adding_packages_to_your_OS_image

and added 
`CORE_IMAGE_EXTRA_INSTALL += "openssh"
`
to the end of my local.conf, the sshd executable appeared in /usr/sbin/ (as it did after I installed via apt-get on my Ubuntu Linux on my laptop)

When I booted the kernel, it automatically started the openssh server, generating all the security-related files and I was able to ssh to it.

Previously I had tried adding openssh to the list of tokens in the IMAGE_INSTALL, but that did not seem to make a difference.

As FYI, I was cross compiling the kernel on x86_64 for ARM32-HF

closed 2020-05-13 by jackfrye