kraj/meta-altera #188: [RFC] Create a consolidated MACHINE for SoCFPGA

rebeccasf, 2020-05-22:
Since stratix10 and agilex are very similar on MACHINE settings and build configurations, I proposed to create a new MACHINE conf named: socfpga.conf as a consolidation for both products and use in future SoCFPGA build. With this consolidation, we can build single image that is able to boot up on multiple devices which in turns ease the maintenance and development of new SoCFPGA in the future. We could still keep stratix10 and agilex machine conf for legacy purposes, but all new products can be consolidated into one machine. Any concern? @dwesterg

kraj, 2020-05-22:
I am all for that, however, how will it reflect on people using either of these machines, does it hide the fact that these machine names are now gone. ? secondly, it perhaps can be addressed with some documentation.

rebeccasf, 2020-05-25:
I think we could give a transition period for those who are using the existing machines before they are removed from the layer. Yes, we would definitely come out with some write up about this move to create more awareness and the introduction of new consolidate machine.

kraj, 2020-05-26:
Sure please propose a patch

rebeccasf, 2020-06-05:
I noticed there are some uboot configurations associated to each machines that needs to clean up as well. I think I have to spend more time on this.

closed 2024-08-19 by rebeccasf