kraj/meta-altera #192: CycloneV Invalid Machine

jackfrye, 2020-06-20:
Hello, I am trying to follow the instructions here.
https://rocketboards.org/foswiki/Documentation/BuildingBootloader#Building_Yocto_Rootfs

It instructs me to run these commands.
```
git clone -b zeus git://git.yoctoproject.org/poky.git
git clone -b master git://github.com/kraj/meta-altera.git
sed -i '/socfpga_cyclone5_de0_nano_soc.dtb/d' meta-altera/conf/machine/cyclone5.conf
source poky/oe-init-build-env ./build
echo 'MACHINE = "cyclone5"' >> conf/local.conf
echo 'IMAGE_FSTYPES = "tar.gz"' >> conf/local.conf
bitbake core-image-minimal
```
I have had success with this flow before. This time, however, I am getting the following issue.
```
$ bitbake core-image-minimal
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    MACHINE=cyclone5 is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.
```
Rather than sifting through files, I was wondering if someone know what the problem might be offhand.

kraj, 2020-06-21:
you missed a step to add it to your layer mix (see the bitbake-layers add-layer step below )
```
git clone -b zeus git://git.yoctoproject.org/poky.git
git clone -b master git://github.com/kraj/meta-altera.git
sed -i '/socfpga_cyclone5_de0_nano_soc.dtb/d' meta-altera/conf/machine/cyclone5.conf
source poky/oe-init-build-env ./build
bitbake-layers add-layer ../meta-altera
echo 'MACHINE = "cyclone5"' >> conf/local.conf
echo 'IMAGE_FSTYPES = "tar.gz"' >> conf/local.conf
bitbake core-image-minimal
```

jackfrye, 2020-06-21:
Thanks. Closing.

closed 2020-06-21 by jackfrye