kraj/meta-altera #194: Building Kernel/RootFS for CycloneV with V4L2 Support

jackfrye, 2020-07-06:
Khem,
What are the configurations to build Video 4 Linux 2 drivers into the kernel and rootfs targeting CycloneV SoC? I have been trying to use https://github.com/altera-opensource/linux-socfpga/blob/socfpga-5.5/arch/arm/configs/socfpga_defconfig to configure and build my kernel. I am cloning and building the kernel outside of the BB layers and Yocto framework.I have been trying to change that configuration definition both manually and with menuconfig, but I cannot seem to get it right. Does the build system automate setting up the kernel configs, and if so can it automate adding in the configs I need to build the V4L2 kernel modules on top of those defined configs? I want the configs to be as close to that config listed above as possible, because I have gotten networking and HPS-FPGA Linux applications working with builds based on that config definition.

kraj, 2020-07-14:
@jackfrye  you have to edit the config using `make menuconfig` once changes are made you can save the modified defconfig with `make savedefconfig` then move newly generated `defconfig` file to `arch/arm/configs/socfpga_defconfig` and you can commit it. if you then use yocto to build the system then apply this delta via a patch

jackfrye, 2020-07-16:
Thanks @kraj . I am going to close the issue.

closed 2020-07-16 by jackfrye