kraj/meta-altera #196: Creating sdcard for DE1SoC

ngohieutruong, 2020-07-16:
Hello Kraj,
I have a problem about SDcard image.
I was run bitbake core-image-minimal successfully.
But I don't known create an image sdcard (such as de1.iso, de1.img,..)? Do you have guide for this problem?
Thanks, have a nice day!

dwesterg, 2020-07-16:
You should be looking for the wic file.

-dalon

kraj, 2020-07-16:
add 
```
IMAGE_FSTYPES += "wic.xz wic.bmap"
```

and then you can dd the wic.xz image

```
$ xzcat <image>-<machine>.wic.xz | sudo dd of=/dev/sdX bs=4M iflag=fullblock oflag=direct conv=fsync status=progress

```

you can also use map image if you flash multiple times a day see how yoe distro does it.

https://github.com/YoeDistro/yoe-distro

ngohieutruong, 2020-07-18:
thank you for replying
[image]
[image]

I have followed your guide. However, it is still not possible to boot Linux on Board. I see in the boot folder on sdcard there is no file.

kraj, 2020-07-20:
if you are having uncompressed wic image then use 
```
sudo dd of=/dev/sdX bs=4M iflag=fullblock oflag=direct conv=fsync status=progress <path/to/wic/image>
```