kraj/meta-altera #217: bitbake  core-image-minimal  gives me an error   on 5.4.64

jjvazha, 2020-12-19:
I was following the https://rocketboards.org/foswiki/Documentation/BuildingYoctoOrAngstromForSoCFPGA
I pulled the  5.4.64 latest HEAD

bitbake  core-image-minimal  gives me an error 

DEBUG: Executing shell function do_shared_workdir
cp: cannot stat ‘Module.symvers’: No such file or directory
WARNING: /home/jojan.vazhaeparampil/workspace/distro/build/tmp/work/cyclone5-poky-linux-gnueabi/linux-altera/5.8+gitAUTOINC+eb446787ab-r0/temp/run.do_shared_workdir.21040:1 exit 1 from 'cp Module.symvers $kerneldir/'
ERROR: Function failed: do_shared_workdir (log file is located at /home/jojan.vazhaeparampil/workspace/distro/build/tmp/work/cyclone5-poky-linux-gnueabi/linux-altera/5.8+gitAUTOINC+eb446787ab-r0/temp/log.do_shared_workdir.21040)

kraj, 2020-12-19:
from your logs it seems that its building 5.8 kernel not 5.4 so you might have to set PREFERRED_VERSION

jjvazha, 2020-12-19:
I am using kraj/meta-altera recepie

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
kraj/meta-altera

Official Altera BSP layer for OpenEmbedded/Yocto Project - kraj/meta-altera
 |

 |

 |

jjvazha, 2020-12-19:
so it is  using kernel
altera-opensource/linux-socfpga

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
altera-opensource/linux-socfpga

Linux development repository for socfpga. Contribute to altera-opensource/linux-socfpga development by creating ...
 |

 |

 |

    On Friday, December 18, 2020, 10:39:18 PM PST, Jojan <jjvazha@yahoo.com> wrote:  
 
  I am using kraj/meta-altera recepie

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
kraj/meta-altera

Official Altera BSP layer for OpenEmbedded/Yocto Project - kraj/meta-altera
 |

 |

 |

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
kraj/meta-altera

Official Altera BSP layer for OpenEmbedded/Yocto Project - kraj/meta-altera
 |

 |

 |

FabianoGK, 2021-06-09:
I do have the same problem:

> ERROR: linux-altera-5.8+gitAUTOINC+eb446787ab-r0 do_shared_workdir: Execution of '/media/Data/Project/build/tmp/work/arria10-poky-linux-gnueabi/linux-altera/5.8+gitAUTOINC+eb446787ab-r0/temp/run.do_shared_workdir.1477800' failed with exit code 1:
> cp: cannot stat 'Module.symvers': No such file or directory

This is my build configuration on Ubuntu 20.04.2 LTS:

```
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "arria10"
DISTRO               = "poky"
DISTRO_VERSION       = "3.0.2"
TUNE_FEATURES        = "arm armv7a vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:74f229160c7f4037107c1dad8f0d02128c080a7e"
meta-oe              = "HEAD:9e60d30669a2ad0598e9abf0cd15ee06b523986b"
meta-altera          = "master:5bb81b77d7bf81a726daec8819fa01c188863f5a"
```

I had to update the linux-altera recipe to linux-altera_5.8 since the previous Linux branches from linux-socfpga have been deleted (deprecated).

For now, I'm using linux-altera-lts version 5.4.104, which compiles successfully. You can add the following to your `layer.conf`:
```
KERNEL_PROVIDER = "linux-altera-lts"
```

kraj, 2021-06-11:
Can you cook a patch and propose here?

FabianoGK, 2021-06-11:
Hi @kraj, actually I was able to successfully compile _linux-altera_5.8_ after upgrading _poky_ and _open-embedded_ layers to _dunfell_. It must be something incompatible with _zeus_ that caused the issue.