kraj/meta-altera #234: UBOOT_EXTLINUX_ROOT

femust, 2022-03-27:
```
ERROR: u-boot-socfpga-1_v2021.04+gitAUTOINC+fcb08e4ed9-r0 do_create_extlinux_config: UBOOT_EXTLINUX_ROOT not defined
ERROR: Task (/home/andrzej/code_workspace/yocto_workspace/andrzej_meta_ros/meta-altera/recipes-bsp/u-boot/u-boot-socfpga_v2021.04.bb:do_create_extlinux_config) failed with exit code '1
```

where the UBOOT_EXTLINUX_ROOT should be setup?

dwesterg, 2022-03-27:
Usually it's defined in the machine configuration.

Dalon

femust, 2022-03-27:
Ok I see, so for some reasons ```cyclone5.conf```did not accept
```
UBOOT_EXTLINUX_ROOT_default ?= "root=/dev/mmcblk0p3"
```

side question since I am still exploring yocto. What does this name specify?

closed 2022-08-25 by kraj