kraj/meta-altera #236: U-Boot env overwrites FAT partition

c-thaler, 2022-08-12:
When building out of the box, the DE10 Nano U-Boot defconfig is configured with "CONFIG_ENV_IS_IN_MMC=y".
This defaults to partition 1, which is the FAT partition as you can see in the WIC file "sdimage-cyclone5-arria5.wks".

That means, saving the U-Boot env may corrupt your FAT data and vice versa.
Either the U-Boot config has to be changed or the WIC file. Which one is the way to go?

kraj, 2022-08-12:
it seems the wic and u-boot are not acting in tandem here. I would think you might do it in either place, perhaps doing it in wic is perhaps yocto specific then you can keep the defconfig of u-boot unmodified.

c-thaler, 2022-08-15:
Ok, then I will add another small partition for the U-Boot environment as partition 1 before the FAT partition. I hope this won't cause any more trouble with other parts of the U-Boot config.

c-thaler, 2022-08-22:
Eventually, I rechecked everything.

CONFIG_ENV_IS_IN_MMC actually defaults to partition 0. Partition 0 contains the MBR and the remaining space (user area) until partition 1 begins, which is the FAT partition in the WIC for Cyclone 5.
The FAT partition is aligned to 1024 kByte (--align 1024). That means, there is enough space for the MBR and the U-Boot env, which starts at 0x4400 and has a length of 0x2000 bytes.

closed 2022-08-22 by c-thaler