kraj/meta-altera #241: meta-altera for Scarthgap

burkulesomesh43, 2024-07-06:
Hi kraj,

I want to use meta altera in Scarthgap. is it supported for the same?
Also could you please suggest suitable azure sdk version for Scarthgap.

this is for new product devlopement and thinking of long term. please suggest if can go ahead with Scarthgap or previous versions.

Thanks,
Somesh

kraj, 2024-07-06:
I would suggest to use https://github.com/altera-opensource/meta-intel-fpga-refdes layer these days

burkulesomesh43, 2024-07-06:
Hi Kraj,

Thanks for your response. Can you please explain more. I'm new to yocto.

meta-intel-fpga-rfdes instead using meta altera?
Seems it also supports kirkstone not scarthgap.

I came to know meta-azure-iot-sdk also doesn't support scarthgap.

Correct me if wrong.

Thanks,
Somesh

kraj, 2024-07-06:
yes, I maintained meta-altera until it was officially supported by Intel so I do not control the roadmap anymore same is for meta-azure-iot-sdk layer, perhaps its better to ask the maintainers of respective layers on their plan and timeline to support scarthgap, I am sure they will support scarthgap eventually since its latest LTS, but the timings may vary.

burkulesomesh43, 2024-07-07:
Hi Kraj,
Thanks for your reply.
Seems meta-intel-fpga-rfdes doesn't support scarthgap. I'll go with kirkstone only.

I came across below repos.
https://github.com/altera-opensource/meta-intel-fpga-refdes/tree/kirkstone
dependencies:
https://github.com/openembedded/meta-openembedded/tree/kirkstone
https://git.yoctoproject.org/meta-intel-fpga/commit/?h=kirkstone&id=74e4dae14953f6c7ea6d49df4f8477bcb71326b2

I'm confused from where to choose all this. could you please guide how to choose this? any official link where I can get all the supported meta's and lib's for kirkstone? I dont' find meta-azure for kirkstone. all looks scattered at different places. I might be missing something

Thanks,
Somesh

kraj, 2024-07-08:
looking at https://github.com/altera-opensource/meta-intel-fpga-refdes/blob/master/conf/layer.conf
it seems their master branch supports both scarthgap and master branches of yocto, so maybe start with master branch of this layer and open Github PRs or Issues for further assistance there.

burkulesomesh43, 2024-07-09:
Thanks kraj. Scathgap doesn't support meta-iotedge(azure).

Kirkstone supports it but in main branch not in stable.

I would like to choose stable or release branch for a product.

closed 2024-07-15 by burkulesomesh43